sannysanoff/SDRPlusPlusBrown #12: logmmse: Increasing AF NR bandwidth sometimes crashes the program

ericek111, 2023-03-19:
Probably a race condition. Happens pretty often. It crashed on me the first time I tried to go from 1 kHz to 2 kHz, but just moving the slider left and right triggers it. Also the AF noise reduction doesn't really do much for my signals.

```
#0  dsp::arrays::diveach (v=..., w=...) at misc_modules/noise_reduction_logmmse/src/arrays.h:219
#1  0x00007fffe009c74d in dsp::logmmse::LogMMSE::logmmse_all (x=..., Srate=Srate@entry=48000, eta=eta@entry=0.150000006, params=params@entry=0x5555564660d8)
    at misc_modules/noise_reduction_logmmse/src/logmmse.h:373
#2  0x00007fffe009d619 in dsp::AFNRLogMMSE::run (this=0x555556465f70) at misc_modules/noise_reduction_logmmse/src/af_nr.h:186
#3  0x00007ffff7e203ed in dsp::block::workerLoop (this=0x555556465f70) at core/src/signal_path/../dsp/channel/../block.h:70
```

sannysanoff, 2024-03-08:
yes, please try NR2.

The first NR is basically deprecated. I will remove it.

sannysanoff, 2024-03-08:
will push the removal of the UI for this NR.

closed 2024-03-08 by sannysanoff