sannysanoff/SDRPlusPlusBrown #13: logmmse: 5 MHz sample rate often crashes with band NR

ericek111, 2023-03-19:
Starting (clicking the start button) with band NR enabled and sample rate of the SDR set > 5 MHz usually causes a crash, but not always. Iif it works, there is heavy stutter even on a Ryzen 3900X.

```
#0  dsp::IFNRLogMMSE::runMMSE at core/src/dsp/types.h:12
#1  dsp::IFNRLogMMSE::run at misc_modules/noise_reduction_logmmse/src/if_nr.h:124
#2  dsp::block::workerLoop at core/src/signal_path/../dsp/channel/../block.h:70
```

sannysanoff, 2024-03-08:
hi, thanks for the report.

this algorithm is very CPU intensive, and I'm afraid 5mhz bandwidth is too much for it.

it's intended for HF SSB applications mostly, and for smaller bandwidth, for sure. I don't even have hardware with 5Mhz width, to play with it.

Meanwhile you could try macos/arm with its accelerated FFT algorithm; maybe it'll be better.

You could investigate it yourself, and if you have some optimization in mind, you're welcome to contribute.

sannysanoff, 2024-03-08:
closing as "i have nothing to add"

closed 2024-03-08 by sannysanoff