sannysanoff/SDRPlusPlusBrown #14: Flickering dropdown under Sinks, and accompanying audible noise

thehans, 2023-07-22:
**Hardware**
- CPU: AMD Ryzen 3900X
- RAM: 32GB 
- GPU: Nvidia GTX 1660
- SDR: rtl-sdr blog v3 local

**Software**
- Operating System: Linux Mint 21.2, Cinnamon desktop
- SDR++: SDR++ Brown v1.1.0-sannysanoff (Built at 16:46:13, Jul 6 2023)

**Bug Description**
Under "Sinks" menu, the dropdown of Audio devices randomly flickers in width.  This causes the down pointing triangle to jump left and right.  

This would be a very minor, easily ignored visual glitch, **except** it also adds a high pitched tick/pop noise to the audio every time the arrow shifts around.

I have also noticed that this arrow consistently jumps to the left when the Stop button is pressed.  And returns to the right edge (for the most part, outside of flickering) when Play button is enabled.

I thought maybe an audio device was being detected intermittently or its description length was changing, but the contents of the drop down do not appear to change when the flickering happens.

This was not an issue for me running the latest SDR++ until recently trying out this Brown fork.

**Steps To Reproduce**
1. Hit play and randomly hear ticks mixed in with audio.
2. Expand the "Sinks" menu to observe flickering which corresponds to audio noise.

**Screenshots**

[image]

sannysanoff, 2024-03-08:
thank you for your report. I will try to reproduce it.

sannysanoff, 2024-03-10:
the resize of ui elements indicated underflow for sound output. It means, audio card wants data / already played data, but there's not enough data provided by sdr++.

This can cause, indeed, by slowness of sdr++ pipeline, which is causes by (as you said), very huge payload of the noise reduction (you applied whole band NR on very wide band).

I removed flickering, i indicate underflow using different means. But you'll have underflow, you'll hear clicking because noise reduction is too slow and does not provide data fast enough, all pipeline is stalled.

sannysanoff, 2024-03-10:
i close the bug, because I changed flickering.

closed 2024-03-10 by sannysanoff