sannysanoff/SDRPlusPlusBrown #15: Console colours

gvanem, 2023-09-02:
It would be nice if the colours for the `core/src/utils/flog.cpp` output would honour the default background console-colour.
It now prints with all black background and in dim colours. On my Windows-console (with a blue background) the start-up now 
shows kinda ugly like this:
[image]

Would IMHO be better like this:
[image]

I've made this patch for myself to get the 2nd console-output above:
``` diff
--- a/core/src/utils/flog.cpp 2023-09-01 09:33:11
+++ b/core/src/utils/flog.cpp 2023-09-02 09:44:56
@@ -31,14 +31,13 @@
         "ERROR"
     };

 #ifdef _WIN32
 #define COLOR_WHITE (FOREGROUND_RED | FOREGROUND_GREEN | FOREGROUND_BLUE)
     const WORD TYPE_COLORS[_TYPE_COUNT] = {
         FOREGROUND_GREEN | FOREGROUND_BLUE,
-        FOREGROUND_GREEN,
-        FOREGROUND_RED | FOREGROUND_GREEN,
-        FOREGROUND_RED
+        FOREGROUND_GREEN | FOREGROUND_INTENSITY,                  // Bright green
+        FOREGROUND_RED | FOREGROUND_GREEN | FOREGROUND_INTENSITY, // Bright yellow
+        FOREGROUND_RED | FOREGROUND_INTENSITY                     // Bright red
     };
 #else
 #define COLOR_WHITE "\x1B[0m"
@@ -159,17 +159,24 @@
             HANDLE conHndl = GetStdHandle(wOutStream);
             if (!conHndl || conHndl == INVALID_HANDLE_VALUE) { return; }

+            static CONSOLE_SCREEN_BUFFER_INFO console_info { 0 };
+            static WORD bg = 0;
+            if (console_info.wAttributes == 0) {
+               GetConsoleScreenBufferInfo (conHndl, &console_info);
+               bg = (console_info.wAttributes & ~7);
+            }
+
             // Print beginning of log line
-            SetConsoleTextAttribute(conHndl, COLOR_WHITE);
+            SetConsoleTextAttribute(conHndl, bg | COLOR_WHITE);
             fprintf(outStream, "[%02d/%02d/%02d %02d:%02d:%02d.%03d] [", nowc->tm_mday, nowc->tm_mon + 1, nowc->tm_year + 1900, nowc->tm_hour, nowc->tm_min, nowc->tm_sec, 0);

             // Switch color to the log color, print log type and
-            SetConsoleTextAttribute(conHndl, TYPE_COLORS[type]);
+            SetConsoleTextAttribute(conHndl, bg | TYPE_COLORS[type]);
             fputs(TYPE_STR[type], outStream);

             // Switch back to default color and print rest of log string
-            SetConsoleTextAttribute(conHndl, COLOR_WHITE);
+            SetConsoleTextAttribute(conHndl, bg | COLOR_WHITE);
             fprintf(outStream, "] %s\n", out.c_str());
 #elif defined(__ANDROID__)
             // Print format string

```

Perhaps it could be added? 

*PS*. I'm a *n00b* when it comes to git and pull requests.

sannysanoff, 2024-03-08:
no worries, added the code.

sannysanoff, 2024-03-08:
done

closed 2024-03-08 by sannysanoff

gvanem, 2024-03-08:
> no worries, added the code.

Where? I do see any change for this in git master.