sannysanoff/SDRPlusPlusBrown #16: [Windows] compiling TCI_server

gvanem, 2023-09-02:
I tried to build the `tci_server.dll` module on Windows (MSVC or clang-cl). That failed miserably due to:
  * the `#include <websocket.h>` line wrongly pulled in WinSDK's `<websocket.h>`. 
  Fixed by `#include "websocket.h"` instead.
  * calling `::read()` and `::close()` for sockets.
  * calling `fcntl()` to set a socket non-blocking.
  * missing `be64toh()` etc.
  * wrong cast in call to `setsockopt()`.
  
My diffs are [here](https://github.com/sannysanoff/SDRPlusPlusBrown/files/12503502/tci_server-diffs.txt) in the hope that it could be added.

*PS*. Could the functions in `core/src/utils/proto/websock.cpp` be used instead?

gvanem, 2024-03-08:
> Basically, this module lacks attention especially in windows environment. 

The same lack of:
 * *Variable Length Arrays* (VLA) on MSVC 
 * `network_source.dll` doesn't work.
 * `kiwisdr_source.dll` works for me, but with somef patches (using `WSAPoll()` etc.)

And why are the network stuff duplicated all over the place; `util/networking.cpp` + `util/net.cpp`? 
E.g. I found no less than 5 calls to `WSAStartup()`! There should be only **one** file to shoe-horn around the Windows/Winsock2/Posix stuff IMHO.

I'll try to come up with patches for the above, but I'm a *n00b* when it comes to C++.

sannysanoff, 2024-04-06:
thank you, you ask right questions.

Sometimes WSAStartup can be found multiple times in code, because loaded shared libraries have their own context, and do not work with sockets without WSAStartup in DLL startup code. Both net.cpp and networking.cpp code belongs to original author. WSAStartup multiple times can be explained there by very same reason. It is possible to trace if one of those calls is used exclusively from modules (who knows?).

Waiting for your patches; thank you.

sannysanoff, 2024-06-12:
kiwisdr source has been fixed since last time, for windows.

sannysanoff, 2024-06-12:
network_source, sorry, no idea.