sannysanoff/SDRPlusPlusBrown #18: Pulseaudio and rtaudio issues?

angrysmiley, 2023-12-02:
Running into two issues ( #13 and #14)  on Ubuntu 22.04 which I believe may be related to Pulseaudio with the rtaudio sink.
I'm aware there was an issue with rtaudio and pulse on the original SDR++ in the past,  however I do believe this issue is unrelated (see below). 

Both issues happen when using pulse no matter device or sample rate.

Issue in #14 is partially mitigated by lowering the sample rate,  the flickering and noise/stutter then almost goes away. 
I don't have any issues here with the original SDR++ 

Both issues are completely mitigated if I either switch to the "New PortAudio Sink" (surprisingly) or disable Pulseaudio (and Pipewire) completely. 

I tried to adjust pulse settings in daemon.conf et al,  but with no luck thus far. 

System is running an AMD Ryzen 7 2700X,  I will see if I can reproduce on another system and/or arch.

sannysanoff, 2024-03-08:
thank you. Probably it will go away when I (or author) moves to native audio drivers.

the cause may be also, because I am using microphone input by default as well. You can try run the application with old driver (not new portaudio), and toggle "Mic" checkbox off. Probably it will change to better. 

Also, I recently optimized microphone processing pipeline, so it may also fix things for you.

Microphone is needed for TRX functionality and voice control (not yet ready).

sannysanoff, 2024-03-10:
please confirm you have any issues when NR is not active.

angrysmiley, 2024-03-26:
Apologies for the late reply. 

I have just tested with the latest source,  same thing happens. 

As soon I enable NR,  it crashes.

n7ihq, 2024-11-19:
I built SDR++ Brown on a laptop with Intel N6000 CPU running Ubuntu 24.04 OS with PulseAudio. I experience stuttering audio with the flashing "Underflow 1" message. After rebuilding with the New Portaudio Sink, the same problem occurs. I have no audio underflows with SDR++ Brown on a Raspberry Pi 4 running Bookworm OS with Pipewire audio.

sannysanoff, 2024-11-26:
hi, please check the original SDR++ software. If problem is same on original project, there's only so much I can do. Author promised native drivers long ago, but I'm afraid he's having long list of more interesting things to do before that... ;(

n7ihq, 2024-11-29:
I installed the original SDR++ version 1.2.1 (Nov. 9 2024) from the deb file sdrpp_ubuntu_noble_amd64.deb. There was no audio stuttering or underflows. Apparently the audio drivers have been updated.

sannysanoff, 2024-11-29:
okay, i will check that out.

n7ihq, 2024-11-29:
With Mic input off (unchecked) the audio stutter is worse and underflow indicator is on continuously. The waterflow update is intermittent.

sannysanoff, 2024-12-04:
Btw, what about portaudio? Is it available in your config instead of rtaudio?

n7ihq, 2024-12-06:
I rebuilt SDR++ Brown using the following options:
cmake .. -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON
Audio settings sysdefault or pulse is clear with no underruns