sannysanoff/SDRPlusPlusBrown #20: Crashing when enabling Baseband NR on ArchLinux

soulnull, 2024-05-01:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Ryzen 5700x
- RAM: 64gb 
- GPU: nVidia 3080 Ti
- SDR: (Remote or local? If remote, what protocol?) RTL_SDR/Airspy HF+ Discovery

**Software**
- Operating System: ArchLinux (latest)
- SDR++: 1.1.0 04-28-2024

**Bug Description**
A clear description of the bug.

Enabling Baseband NR plays for approximately 1 second, then freezes and crashes. Running in terminal produces this error:

Sampling piece... srate=912000 Slen=18240 nFFT=36480
/usr/include/c++/13.2.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = int; _Alloc = std::allocator<int>; reference = int&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)

Running in Ubuntu 22.04 Distrobox allows Baseband NR to work, but causes very bad audio stuttering. I'm just trying to identify where to even look to fix this, as it's seemingly an Arch issue (issue with the latest g++ perhaps?).. If you point me in a direction, I'd be grateful.

**Steps To Reproduce**
1. Start SDR++ on Archlinux
2. Enable Baseband NR
3. Crash

**Additional info**
Add any other relevant information.

sannysanoff, 2024-05-14:
please check your cpu usage. It can be your CPU is not powerful enough for given sample rate (almost 912 ksps). Try to reduce sample rate. Baseband NR is expensive operation. Probably I will need to add safety check that will skip Baseband NR if it consumes too much resources to proceed.

Of course, there's a chance of a bug, however it works quote ok for long time. Please attempt on other CPU and post your results here.

sannysanoff, 2024-05-14:
based on your exception information, however, I would like to reproduce it. Please turn off baseband NR, record around 10 seconds of baseband using recorder, and attach wav here or somewhere else. That would help.

sannysanoff, 2024-05-14:
just for reference, it adds +30% of CPU usage on Mac M2 for bandwidth=2.4MSPS (rtl sdr) using release build.

soulnull, 2024-05-14:
I appreciate your time, thank you.

I have compiled with RelWithDebInfo. I included bt, but I also included more of the log prior to as well.
```
[14/05/2024 15:24:12.972] [INFO] New DSP samplerate: 1000000.000000 (source samplerate is 1000000.000000)
[14/05/2024 15:24:12.972] [INFO] SpyServerSourceModule 'SpyServer Source': Menu Select!
[Thread 0x7fffc50006c0 (LWP 239435) exited]
[New Thread 0x7fffc50006c0 (LWP 239635)]
[Thread 0x7fffcca006c0 (LWP 239434) exited]
[New Thread 0x7fffcca006c0 (LWP 239636)]
[Thread 0x7fff7ea006c0 (LWP 239634) exited]
[Thread 0x7fff85e006c0 (LWP 239629) exited]
[Resamp] predec: 32, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[New Thread 0x7fff85e006c0 (LWP 239637)]
[Thread 0x7fff84a006c0 (LWP 239630) exited]
[Resamp] predec: 32, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[New Thread 0x7fff84a006c0 (LWP 239638)]
[Resamp] predec: 4, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Thread 0x7fffb74006c0 (LWP 239632) exited]
[Thread 0x7fffb6a006c0 (LWP 239633) exited]
[Thread 0x7fffb7e006c0 (LWP 239631) exited]
[New Thread 0x7fffb6a006c0 (LWP 239639)]
[New Thread 0x7fffb74006c0 (LWP 239640)]
[New Thread 0x7fffb7e006c0 (LWP 239641)]
[New Thread 0x7fff7ea006c0 (LWP 239642)]
[Thread 0x7fff7ea006c0 (LWP 239642) exited]
[Thread 0x7fff85e006c0 (LWP 239637) exited]
[Resamp] predec: 32, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[New Thread 0x7fff85e006c0 (LWP 239643)]
[Thread 0x7fff84a006c0 (LWP 239638) exited]
[Resamp] predec: 32, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[New Thread 0x7fff84a006c0 (LWP 239644)]
[Resamp] predec: 4, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Thread 0x7fffb7e006c0 (LWP 239641) exited]
[Thread 0x7fffb74006c0 (LWP 239640) exited]
[Thread 0x7fffb6a006c0 (LWP 239639) exited]
[New Thread 0x7fffb7e006c0 (LWP 239645)]
[New Thread 0x7fffb74006c0 (LWP 239646)]
[New Thread 0x7fffb6a006c0 (LWP 239647)]
[New Thread 0x7fff7ea006c0 (LWP 239648)]
[14/05/2024 15:24:12.975] [INFO] New DSP samplerate: 1000000.000000 (source samplerate is 1000000.000000)
[14/05/2024 15:24:12.975] [INFO] rawFFTS: 0x0x5555560fc530, length 0
[Thread 0x7fffb74006c0 (LWP 239646) exited]
[Thread 0x7fffb7e006c0 (LWP 239645) exited]
[Thread 0x7fffb6a006c0 (LWP 239647) exited]
[New Thread 0x7fffb6a006c0 (LWP 239649)]
[New Thread 0x7fffb74006c0 (LWP 239650)]
[New Thread 0x7fffb7e006c0 (LWP 239651)]
[Thread 0x7fffb74006c0 (LWP 239650) exited]
[Thread 0x7fffb7e006c0 (LWP 239651) exited]
[Thread 0x7fffb6a006c0 (LWP 239649) exited]
[New Thread 0x7fffb7e006c0 (LWP 239652)]
[New Thread 0x7fffb74006c0 (LWP 239653)]
[New Thread 0x7fffb6a006c0 (LWP 239654)]
[Thread 0x7fffb6a006c0 (LWP 239654) exited]
[Thread 0x7fffb74006c0 (LWP 239653) exited]
[Thread 0x7fffb7e006c0 (LWP 239652) exited]
[New Thread 0x7fffb6a006c0 (LWP 239655)]
[New Thread 0x7fffb74006c0 (LWP 239656)]
[New Thread 0x7fffb7e006c0 (LWP 239657)]
[14/05/2024 15:24:12.982] [INFO] SpyServerSourceModule 'SpyServer Source': Tune: 1220237.793626!
[14/05/2024 15:24:12.982] [INFO] Running post-init for Airspy Source
[14/05/2024 15:24:12.982] [INFO] Running post-init for AirspyHF+ Source
[14/05/2024 15:24:12.982] [INFO] Running post-init for Audio Sink
[14/05/2024 15:24:12.982] [INFO] Running post-init for Audio Source
[14/05/2024 15:24:12.982] [INFO] Running post-init for FT8/FT4 Decoder
[14/05/2024 15:24:12.982] [INFO] Running post-init for File Source
[14/05/2024 15:24:12.982] [INFO] Running post-init for Frequency Manager
[14/05/2024 15:24:12.982] [INFO] Running post-init for HackRF Source
[14/05/2024 15:24:12.982] [INFO] Running post-init for Hermes Source
[... 81 lines cut ...]

Thread 222 "block:IFNRLogMM" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff1ea006c0 (LWP 239686)]
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
Downloading source file /usr/src/debug/glibc/glibc/nptl/pthread_kill.c
44            return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;                                                                                                                                                            
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff76a8eb3 in __pthread_kill_internal (threadid=<optimized out>, signo=6) at pthread_kill.c:78
#2  0x00007ffff7650a30 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff76384c3 in __GI_abort () at abort.c:79
#4  0x00007ffff68d2d60 in std::__glibcxx_assert_fail (file=<optimized out>, line=<optimized out>, function=<optimized out>, condition=<optimized out>) at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/assert_fail.cc:41
#5  0x00007fffdfbdc62f in ??? () at /usr/lib/sdrpp/plugins/noise_reduction_logmmse.so
#6  0x00007fffdfbe65e7 in dsp::IFNRLogMMSE::run() () at /usr/lib/sdrpp/plugins/noise_reduction_logmmse.so
#7  0x00007ffff78f019a in dsp::block::workerLoop() () at /usr/lib/libsdrpp_core.so
#8  0x00007ffff68e0c84 in std::execute_native_thread_routine (__p=0x5555a966c770) at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/thread.cc:104
#9  0x00007ffff76a6ded in start_thread (arg=<optimized out>) at pthread_create.c:447
#10 0x00007ffff772a0dc in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
(gdb) 
```

As for baseband recording:
https://deadfrog.org/baseband_1220238Hz_15-22-09_14-05-2024.wav

[... cut ...] , as I've had to do some hacky things for other stuff in the past and this may be the root cause of some other issues... So by all means, if this is my fault, you have permission to admonish me for wasting your time, preferably upon revealing what you're seeing that could indicate hacky shenanigans (why does it say c++11? i don't have c++11, only 12 and 13... related? Investigating)... But that Baseband NR is amazing on my other devices, an absolute game changer that I hope to use on my big rig.

sannysanoff, 2024-05-18:
inside your stack trace it is unfortunately impossible to detect where exactly the out of bounds error happened. For some reason, debug information is missing in your stack traces. For example, 

#4  0x00007ffff68d2d60 in std::__glibcxx_assert_fail (file=<optimized out>, line=<optimized out>, function=<optimized out>, condition=<optimized out>) at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/assert_fail.cc:41
#5  0x00007fffe0bdc62f in ??? () at /usr/lib/sdrpp/plugins/noise_reduction_logmmse.so

   ^^^^^^^^^^^^^^^^^^^^

this address is not translated to line number or symbol. Looks like you're running release version. Please build with

cmake -DCMAKE_BUILD_Type=RelWithDebInfo

Ensure it has debug info.

In general, it has a lot of array (vector) access using operator[] to check them all. I tried to simulate slow CPU by adding various delay loops, but it worked in my case all time.

HOWEVER, I added slow cpu check, NR will switch off automatically when CPU usage is too high for 2 iterations in row. 

Hope it doesn't spoil all enjoyment.

sannysanoff, 2024-05-18:
also I optimized NR by 10-15 %, but looks lit it was already in your last build that you tested.

soulnull, 2024-05-21:
That's definitely interesting, as I have added it to the PKGBUILD and rebuilt.

I'm going to wipe out that whole /usr/lib/sdrpp folder and try again, maybe it's a stale old library.

soulnull, 2024-05-21:
User error: -O3 override causes crashes, default -O2 does not. I will close this.

closed 2024-05-21 by soulnull