sannysanoff/SDRPlusPlusBrown #21: SDR++Brown Crashes on startup on macos

lscalpati, 2024-05-29:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 2.5 GHz Dual-Core Intel Core i5
- RAM: 16 GB 1600 MHz DDR3
- GPU: Intel HD Graphics 4000 1536 MB
- SDR: Local

**Software**
- Operating System: MacOS 14.4.1
- SDR++: Rolling release 5/28/2024

**Bug Description**
App crashes on launch
SDR++Brown quit unexpectedly.
Click Reopen to open the application again. Click Report to see more detailed information and send a report to Apple.

**Steps To Reproduce**
double click the app file

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

lou@ham4 MacOS % ./sdrpp -c
2024-05-28 22:33:27.699 sdrpp[17719:410238] MACOS: Already authorized
[28/05/2024 22:33:27.699] [INFO] SDR++Brown v1.1.0-sannysanoff
Unknown argument
-a --addr		Server mode address
   --autostart		Automatically start the SDR after loading
-h --help		Show help
-p --port		Server mode port
-r --root		Root directory, where all config files are stored
-s --server		Run in server mode
-x --temp		Temp directory, where all temporary files are stored
zsh: segmentation fault  ./sdrpp -c

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

lscalpati, 2024-05-29:
I removed the -c parameter since it was not found and reran without any parameters:

lou@ham4 MacOS % ./sdrpp   
2024-05-28 22:36:35.088 sdrpp[17805:411967] MACOS: Already authorized
[28/05/2024 22:36:35.089] [INFO] SDR++Brown v1.1.0-sannysanoff
Current path = "/Applications/Apps-HamRadio/SDR++Brown.app/Contents/MacOS"
[28/05/2024 22:36:35.090] [INFO] Loading config from: /Users/lou/Library/Application Support/sdrpp-brown (path /Applications/Apps-HamRadio/SDR++Brown.app/Contents/MacOS)
[28/05/2024 22:36:35.091] [WARN] ConfigManager locked, waiting...
2024-05-28 22:36:35.256 sdrpp[17805:411967] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.
[28/05/2024 22:36:35.338] [ERROR] Glfw Error 65545: NSGL: Failed to find a suitable pixel format
zsh: segmentation fault  ./sdrpp
lou@ham4 MacOS %

sannysanoff, 2024-06-12:
Thanks for your report!
Does original version work on same hardware configuration?

sannysanoff, 2024-06-15:
I just downloaded original SDR++ on mac intel, myself. I observe same crash. Something broke in orig author's code.

However, when you launch it from the terminal, like 

username$ Downloads/SDR++.app/Contents/MacOS/sdrpp

then it starts up normally.

I will see what I can do.

lscalpati, 2024-06-15:
Yes, that starts the app.  Have you filed an issue on the original
product?  If not I will.  From my perspective this work around is
something I can script a round and should not take away from the fine work
you are doing on the brown release.  Thanks

sannysanoff, 2024-10-04:
Thank you. If original author patches that, hopefully it will propagate here as well.

sannysanoff, 2024-10-04:
considering as "not mine"

closed 2024-10-04 by sannysanoff