sannysanoff/SDRPlusPlusBrown #22: Great project, some requests & issues

Sultan-papagani, 2024-06-12:
-Device-
Galaxy S23
Android 14
OneUI 6.0

-Description-
Thank you for this awesome fork, it really helps on mobile devices. but its not stable, FT8/FT4 decoder causes crashes, When you open the app again Waterfall's min-max slider is in the wrong position (it resets to max value), Noise reduction sometimes crashes the app, 

-Suggestion-
1)"Audio NR2 Radio 1.0" never  sounds good, there should be adjustable bandpass, bandstop filters for better audio.
 2) AM bandwidth's max value should be increased, its not enough

sannysanoff, 2024-06-15:
Thanks for the report

About FT8 - can you give any insights? How long does decode take (should be quick). I never experienced crashes in FT8... 

Minmax with waterfall - gonna check. Do you close/open app completely, or switch to other app and then return?

About noise reduction: please provide your CPU usage (it displays the figure besides the checkbox), if you're talking about the Band NR.

About Audio NR2 Radio, please provide examples (wav, mp4 recording etc on and off state), because I would happy to know how it sounds for other people. I don't know why bandpass would help there, because I'm novice. Please explain your suggestion.

About max AM bandwidth: probably this maximum is set in the original software. Why it's not enough for you? Please provide the baseband recording where you have such wide signal.

Thanks a lot.

Sultan-papagani, 2024-09-25:
closed due to inactivity