sannysanoff/SDRPlusPlusBrown #23: Missing remote options on SDR++ Server

davmen2, 2024-06-15:
In the latest rolling version, when connecting to a remote SDR++ server, the remote device configuration options, bandwidth, gain, etc. have disappeared.
[image]

sannysanoff, 2024-06-15:
Do you connect to same version of sdr++brown on server?

davmen2, 2024-06-15:
no... the sdrpp server is not the brown version, it is the v1.2 Alexander Rouma and installed on a raspberry pi5 and the client is the version for windows 64b

davmen2, 2024-06-15:
Dont worry, i use the brown version because it is very light and i love the modules like Noise Reduction Logmmse

ThorAlex87, 2025-06-12:
I wanted to check out SDR++ brown for the scanner and noise reduction features, but I ran into this issue. It also seems I can only install one of them at a time as my laptop (Ubuntu 25.04) seems to treat both SDR++ and SDR++ brown as the same program and I can only replace one with the other. I'd like to either run both programs on my servers, or somehow make SDR++ brown work with SDR++ server.

The reason for wanting a solution (other than just switching server program) is I run multiple SDR++ servers on two different rpi 4's with two servers running on each, and access them from the same laptop. One (or sometimes two) of the servers are accessed with satdump for playing around with sattelite reception, and I assume satdump would have similar compatibility issues if I where to connect it to an SDR++ brown server.  I'm not great with the command line so I'd rather not mess with anything on the server side before asking for advice...