sannysanoff/SDRPlusPlusBrown #24: Consider adding digital decoding plugin? DMR and Tetra Support

Wonkaski, 2024-07-17:
**Feature description**
https://github.com/cropinghigh/sdrpp-tetra-demodulator
https://github.com/cropinghigh/sdrpp-vhfvoiceradio

See these two fantastic plugins for SDR++, really helpful and often-used features, should be added to this this project to make it even better!!

sannysanoff, 2024-09-02:
@Wonkaski hi, vhfvoiceradio is ALREADY added, it also has different name ;) 
tetra not added, maybe later, same way.

Wonkaski, 2024-09-02:
@sannysanoff Thanks for replying and your awesome work! I think I am using the MacOS Arm version from latest July build, and I did not see the integration of vhfvoiceradio (which obviously integrated into "radio" part, because I checked the Windows x64 version, it is there). I don't really know much about programming and how to integrate these plugins, so I would probably not able to know if it is just build issue or you intentionally did not put it in the MacOS version because of technical difficulties or something else. If I know how to build or program these things, I would definitely help by my best.

sannysanoff, 2024-10-04:
look for ch_extravhf_decoder