sannysanoff/SDRPlusPlusBrown #28: Kiwisdr missing as source

loewal, 2025-01-09:
**Hardware**
- CPU:  Mac Mini M2
- RAM: 8GB
- GPU: 
- SDR: local

**Software**
- MacOs Sequioa 15.3
- SDR++: v1.2.1 built 30-12-2024

**Bug Description**
Kiwisdr not visible as source; during startup this is reported as missing

**Steps To Reproduce**
1. Start SDRpp Brown
2. Select source
3. no Kiwi; also not under modules

How can I add it?
Thanks!

loewal, 2025-01-09:
I compiled myself and copied kiwisdr_source.dylib to the plugins folder. First tried it in the original SDRpp for a test.
Now I get:

[09/01/2025 07:41:54.000] [ERROR] Couldn't load /Applications/Korte_Golf/SDR++.app/Contents/Resources/../Plugins/kiwisdr_source.dylib: dlopen(/Applications/Korte_Golf/SDR++.app/Contents/Resources/../Plugins/kiwisdr_source.dylib, 0x0005): tried: '/Applications/Korte_Golf/SDR++.app/Contents/Resources/../Plugins/kiwisdr_source.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Applications/Korte_Golf/SDR++.app/Contents/Resources/../Plugins/kiwisdr_source.dylib' (no such file), '/Applications/Korte_Golf/SDR++.app/Contents/Resources/../Plugins/kiwisdr_source.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/Applications/Korte_Golf/SDR++.app/Contents/Plugins/kiwisdr_source.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Applications/Korte_Golf/SDR++.app/Contents/Plugins/kiwisdr_source.dylib' (no such file), '/Applications/Korte_Golf/SDR++.app/Contents/Plugins/kiwisdr_source.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))

Is this also what happened with your compilation?

loewal, 2025-01-09:
Found solution:
Copied kiwisdr_source.dylib to the pluginsdirectory of the Intel-version.

This works!!
Is it possible to compile an ARM-version with compatible plugins? Not only Kiwi, but several other plugins were not found...Perseus, Pluto...etc.

loewal, 2025-01-11:
Hi,
Perhaps a neater solution which works fine:
ln -s /usr/local/lib/libsdrplay_api.so.3 /Applications/Korte_Golf/"SDR++Brown.app"/Contents/Frameworks
ln -s /usr/local/lib/kiwisdr_source.dylib /Applications/Korte_Golf/"SDR++Brown.app"/Contents/Plugins

FerTouny, 2025-06-14:
> Hi, Perhaps a neater solution which works fine: ln -s /usr/local/lib/libsdrplay_api.so.3 /Applications/Korte_Golf/"SDR++Brown.app"/Contents/Frameworks ln -s /usr/local/lib/kiwisdr_source.dylib /Applications/Korte_Golf/"SDR++Brown.app"/Contents/Plugins

Hi, could you please explain how to fix the issue?

SA7BNT, 2025-06-15:
> Found solution: Copied kiwisdr_source.dylib to the pluginsdirectory of the Intel-version.
> 
> This works!! Is it possible to compile an ARM-version with compatible plugins? Not only Kiwi, but several other plugins were not found...Perseus, Pluto...etc.

Hey,
would you share your version for Mac Intel including Kiwi online?

Best 
Tim

guero-cmd, 2025-06-17:
Is it possible to compile just the module kiwisdr_source.dylib from SDRPlusPlusBrown/tree/master/source_modules/kiwisdr_source/src/main.cpp?

If so, would it be >make and then >make install ?