sannysanoff/SDRPlusPlusBrown #35: BUG: Map shows Kiwis not correct

DL4QB, 2025-06-11:
Running Windows 11 here, latest version. Looks nice so far, my very first tests, but the map doe not show the KIWIs at the right place.

[image]

sannysanoff, 2025-06-11:
file is valid, strange. If you want to provide feedback, monitor this thread, I will add some logging in next release. It will take some time. You'll run updated version and hopefully we'll get something.

sannysanoff, 2025-06-12:
There's new windows version. Check it. If error persists, plz use "-c" launch argument for console output, please quote logs from there, related to the lat/long parsing.

derLoosi, 2025-06-12:
Hey, same problem for me, Windows 11 24H2 Build 26100.4202 using the latest version from 3hours ago. If i run it using -c i see the following in the console:

```
[12/06/2025 11:40:43.000] [INFO] Running post-init for WebSDR View
[12/06/2025 11:40:43.000] [INFO] Ready.
[12/06/2025 11:40:49.000] [WARN] Parsing geo coordinates: "(29.013017, -82.453791)" => 29,000000,0,000000
[12/06/2025 11:40:49.000] [WARN] Parsing geo coordinates: "(46.593936, -114.028000)" => 46,000000,0,000000
[12/06/2025 11:40:49.000] [WARN] Parsing geo coordinates: "(52.763229, 41.502566)" => 52,000000,0,000000
[repeats a few hundret times]
[12/06/2025 11:40:50.000] [INFO] Parsed 858 servers
```

sannysanoff, 2025-06-12:
i researched and pushed fix, in 30 minutes it will be built, enjoy.

closed 2025-06-12 by sannysanoff