sannysanoff/SDRPlusPlusBrown #39: Frequency Manager: DSD mode is not recognised in memory item after SDR++ is restarted

mrdonkersley, 2025-06-12:
**Hardware**
- CPU: 
- RAM:
- GPU: 
- SDR: SDRPlay RSP1 local

**Software**
- Operating System: Windows 11 x64,
- SDR++: Version 1.2.1-brown built at 06:35:01, Jun 12 2025

**Bug Description**
I saved a frequency with the mode DSD, in frequency manager. This shows as 4865 if the frequencymanager.config file is opened. Whilst the program is running the memory and mode is retained and functions normally if clicked. If the program is closed and reopened, then the memory name, freq, bandwidth are shown in the GUI  but the mode is not, is shown as blank and defaults to  FM if the memory is selected. The entry in frequencymanager.config is unchanged, though, and still shows 4865

**Steps To Reproduce**
1. ...as above
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

Bfreak, 2025-11-17:
Also checking in with this bug, very annoying. Only fix for me is to restart.

Bfreak, 2025-11-18:
@mrdonkersley I found a goofy workaround, and possibly a part of the problem! Make another list with an FM frequency on it, and then modify your frequency_manager_config.json file so that the scanner selected list is the new one. Then when you restart, your main list will still be the default, and still assign FM to the frequncies, but if you switch to the new list, and then back to your original list, they will have all switched back to DSD.

FabianInostroza, 2026-04-26:
I have fixed this in my "freq_manager_test" branch, but after commit f75cc5d5bc70d105f4ed3e8952eb137476e4a5fe DSD or DSD OLD are not shown in the GUI.
Applying the changes on commit 36eb6fd7048d06b7b2aead7b247c601d9642f8e6 works fine