sannysanoff/SDRPlusPlusBrown #50: Package file conflicts, mbelib and itpp

edward-p, 2025-07-04:
Some "decoder_modules" will download and build mbelib, itpp from source. When execute `make install`, the mbelib and itpp files will also be installed. Which will cause file conflicts between packages. For Linux builds, using installed libraries by package managers are preferred.

alch-emi, 2025-12-06:
Chiming in to say I ran into this issue while trying to set up a build on Nix.  When building on Nix, this doesn't manifest as a conflict, but instead causes the build to fail, since network access is blocked while running builds.  It would be nice if the download/build was only attempted if the library isn't already installed on the system.

sannysanoff, 2025-12-29:
guys, please provide safe fallback, I'll accept PR.

closed 2025-12-29 by sannysanoff