sannysanoff/SDRPlusPlusBrown #56: cant find BLAS

gariac, 2025-07-21:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: AMD Ryzen 7840u
- RAM:96g
- GPU: 780M
- SDR: (Remote or local? If remote, what protocol?)rtlsdr 

**Software**
- Operating System: debian 13
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**

     -- A library with BLAS API not found. Please specify library location.
     -- A library with BLAS API not found. Please specify library location.
    -- LAPACK requires BLAS
    -- A library with LAPACK API not found. Please specify library location.

It appears there are a few ways to set up BLAS.

https://wiki.debian.org/DebianScience/LinearAlgebraLibraries

I have installed:
libblas3  libblas-dev
libmkl-rt  libmkl-dev)
libblis-dev

sannysanoff, 2025-11-07:
hello, cmake should automatically detect system-installed packages, there should not be special handling. What is your operating system (exactly), cmake versions etc

sannysanoff, 2025-12-28:
CMake has a built-in flag specifically for this:

cmake -S . -B build --debug-find

or if you already have a build directory:

cd build
cmake .. --debug-find

This prints every search path, library name, and failure related to find_package().

closed 2025-12-28 by sannysanoff